EttusResearch/uhd #819: How to use enable_rx_timestamps()?

tomwts, 2024-12-05:
I want Tx radio begins transmitting about 1/4 sec after Rx radio starts streaming samples to DDC and dmaFIFO to simulate a RF path delay. After enable_rx_timestamps() is added to my app, the Tx radio block no longer sends data. Instead, I got a bunch of "L" errors printed on the console. I thought by setting a 1/4 sec delay in the issue_stream_cmd(), it would be sufficient to cause the received samples saved in the FIFO until Tx radio is unblocked, but it didn't work.

Some of the UHD 4.7 release example app like rfnoc_radio_loopback has rx-timestamps command line option, but enabling it resulted in getting numerous "L" errors and the Tx radio block couldn't send any data.

How enable_rx_timestamps() works and how to use it properly?

Thanks,

mbr0wn, 2024-12-13:
@tomwts The Rx radio puts the receive time as timestamps on the packets, but the Tx radio will interpret timestamps as a request to transmit a sample at that time. So if timestamps are enabled, by definition, you will always have lates (and late packets aren't transmitted). Honestly I am pondering removing `--rx-timestamps` from `rfnoc_radio_loopback`, as you always have to leave it off for that example.

What you want/need is a block that adds 250ms worth of time to the timestamps, but we don't have that in UHD.
So the answer to your question is "you always turn off timestamps unless you have a way to modify timestamps in-flight".

I'm going to close this issue, but I acknowledge that while I've answered the question,  I haven't helped you solve the 250ms delay. Please open a feature request if you think you want to.

closed 2024-12-13 by mbr0wn

tomwts, 2024-12-16:
Hi Martin, thanks for your explanation. May I ask you a couple of questions?
I have been working on a custom Tx radio block that can handle delayed sample packets. Is the 64bit timestamp value stored in the sample packet header in Big- or Little-Endian format?

Inside radio_tx_core.v, there are debug function $display calls. How can I enable the debug output in the image build and what tool can be used to collect such FPGA debug output? I appreciate it if you can provide me a link to the related documentation.

Thanks,

mbr0wn, 2024-12-19:
> @mbr0wn <https://github.com/mbr0wn> Hi Martin, thanks for your

Hey @tomwts, I recommend you use the mailing list for actual discussions,
so we can focus on bug reports here. It sounds like you're asking about
generic FPGA development, which is a big field. The debug output is used
for the testbench. If you want to get involved in FPGA development at the
HDL level, I suggest you get a book or two on Verilog and SystemVerilog,
and start poking around in the code.