EttusResearch/uhd #828: Unable to add VHDL sources to RFNoC block

zacaric, 2025-01-08:
## Issue Description ##

I have some VHDL code that I would like to use in the top system verilog file for the block (`rfnoc_block_<block_name>.sv`), but when I try to add it to the `RFNOC_BLOCK_<block_name>_SRCS`, it has the following error:
```bash
CMake Error at fpga/<oot_name>/rfnoc_block_<block_name>/cmake_install.cmake:46 (file):
  file INSTALL cannot find
  "/workspaces/rfnoc-<oot_name>/fpga/<oot_name>/rfnoc_block_<block_name>/add8bit.v":
  No such file or directory.
Call Stack (most recent call first):
  fpga/<oot_name>/cmake_install.cmake:51 (include)
  cmake_install.cmake:48 (include)
```
`add8bit.v` should be `add8bit.vhd`

## Setup Details ##

`UHD 4.7.0.0-149-g635ad362` in a `Ubuntu 22.04` docker image.

## Expected Behavior ##

Able to use VHDL, Verilog, or System Verilog 

## Actual Behaviour ##

Only can use System Verilog for the top module and Verilog for all other modules

## Steps to reproduce the problem ##

1. Create a file in `fpga/<oot_name>/rfnoc_block_<block_name>` called `add8bit.vhd` and create a vhdl module called `add8bit`. 
2. Add the file to `fpga/<oot_name>/rfnoc_block_<block_name>/Makefile.src` file like so:
```Makefile
RFNOC_BLOCK_<block_name>_SRCS += $(addprefix $(dir $(abspath $(lastword $(MAKEFILE_LIST)))), \
rfnoc_block_<block_name>.sv \
noc_shell_<block_name>.sv \
add8bit.vhd \
)
```

## Additional Information ##

mbr0wn, 2025-01-09:
@zacaric Thanks for reporting. This is indeed an issue. Looks like the regex for identifying files in Makefiles.srcs is borked (and the CMake regex engine makes it easier for this mistake to occur).

A fix is on the way -- I just can't promise that it will make it into the upcoming release.

mbr0wn, 2025-01-09:
FYI here's a fix:

```
diff --git a/host/examples/rfnoc-gain/CMakeLists.txt b/host/examples/rfnoc-gain/CMakeLists.txt
index 2533f4155..8ecc49444 100644
--- a/host/examples/rfnoc-gain/CMakeLists.txt
+++ b/host/examples/rfnoc-gain/CMakeLists.txt
@@ -151,7 +151,11 @@ macro(RFNOC_REGISTER_BLOCK_DIR)
     message(STATUS "Registering RFNoC block: ${_blk_name}")
     file(READ ${CMAKE_CURRENT_SOURCE_DIR}/Makefile.srcs _makefile_srcs)
     list(APPEND _block_src_files "Makefile.srcs")
-    string(REGEX MATCHALL "[a-z0-9_]+\\.(s?v|vhdl?)" _src_files ${_makefile_srcs})
+    # Note: CMake uses an old regex engine that does not (among other things)
+    # support word boundaries. When changing this regex, make sure that it
+    # full captures intended file names. For example, file.v is a subset of
+    # file.vhd, so the regex should match file.vhd first.
+    string(REGEX MATCHALL "[a-z0-9_]+\\.(vhdl?|s?v)" _src_files ${_makefile_srcs})
     foreach(_src_file ${_src_files})
         string(STRIP "${_src_file}" _src_file})
         list(APPEND _block_src_files "${_src_file}")
```

zacaric, 2025-01-09:
That fixed it. Thanks so much for the quick response and help!

closed 2025-01-09 by joergho