EttusResearch/uhd #835: Undefined Symbol in Custom OOT Library

zacaric, 2025-02-10:
## Error
```
Generating: "/workspaces/untitled.py"

Executing: /usr/bin/python3 -u /workspaces/untitled.py

Traceback (most recent call last):
  File "/workspaces/untitled.py", line 22, in <module>
    from gnuradio import rfnoc_my_oot
  File "/usr/local/lib/python3.10/dist-packages/gnuradio/rfnoc_my_oot/__init__.py", line 18, in <module>
    from .rfnoc_app_specific_python import *
ImportError: /usr/local/lib/libgnuradio-rfnoc_my_oot.so.1.0.0: undefined symbol: _ZTIN2gr3uhd11rfnoc_blockE

>>> Done (return code 1)
```

Also when building the image for the image, I get the following error that pops up. I also get it when running `uhd_config_info --version`
```
Exception in static block load_modules
  EnvironmentError: OSError: dlopen failed to load "librfnoc-app_specific.so"
```

## How to Produce
I just created a normal OOT module, added registers to the HDL code, then add APIs to those registers using rfnoc-gain as an example. I have 7 registers that I'm using.

## Versions
UHD: `UHD 4.8.0.0-0-g308126a4`
GNURadio: `v3.11.0.0git-912-g58b3b649`

## Additional Information

I am not sure what else to add to this, but will be glad to share as much as my work is willing to let me share.

zacaric, 2025-03-17:
Unfortunately my work pulled funding for what I was trying to do. How many registers are you using? Before they ended it I was going to try and see if removing all of the registers would fix it and if that was causing the issue.

SJ-Innovation, 2025-03-17:
This isnt a register/not issue i can tell you that for free. That symbol its trying and failing to find is the base class of the rfnoc block from which any gnuradio hooks to rfnoc blocks derive. Its weird that it compiles & installs just fine, but then fails to load, leading me to believe its a loader/rpath related issue. Disconnecting the block entirely from the underlying controller still demonstrates this error, apparently even just having a source file which refers to rfnoc breaks things. Annoyingly though the gr-rfnoc_gain example works though, which is confounding because theres visibly no difference. :/ Clearly there is, I'm just too blind to see it. 

Sorry to hear about your funding btw.

zacaric, 2025-03-17:
That is good to know for the future if we can get another customer to pay for some extra functionality

SJ-Innovation, 2025-03-18:
FYI, found the issue at least for me. Any projects gr- block module needs to also link against uhd (obvious, and thought it was already done, just missed adding it in one CMakeLists.txt).

gr-<module>/libs/CMakeList.txt: Line ~32
target_link_libraries(gnuradio-<module> gnuradio::gnuradio-runtime gnuradio::gnuradio-uhd) 

Closing.

zacaric, 2025-03-20:
I tried really quick and that fix didn't work for me