EttusResearch/uhd #840: build failed uhd 4.8.0.0 with dpdk installed in dedicated path (not /usr/)

KarimH22, 2025-02-27:
Hello

I tried to build uhd 4.8.0.0 and strangely the DPDK_CFLAGS are not added to build transport

I got 
```
uhd-4.8.0.0/host/lib/include/uhdlib/transport/dpdk/common.hpp:23:10: fatal error: rte_arp.h: No such file or directory
   23 | #include <rte_arp.h>
      |          ^~~~~~~~~~~
```

You pushed in host/lib/CMakeLists.txt 
`target_compile_options(uhd PRIVATE ${DPDK_CFLAGS})`
but building with make VERBOSE=1 
I dont see DPDK_CFLAGS

It is better to keep in 
host/lib/transport/CMakeLists.txt 

`include_directories(${DPDK_INCLUDE_DIRS})`

NI-LAm, 2025-02-28:
How did you install cmake? `/opt` look like it is not the DPDK package of your distribution. If the CFLAGS are empty there is something wrong with your DPDK installation. This is probably something to be asked in a DPDK forum.

KarimH22, 2025-02-28:
cmake is installed with apt
dpdk was build manually with --prefix=/opt/dpdk/ , and then `ninja install`
 so that it is not installed in the usual folder /usr/

KarimH22, 2025-02-28:
Hello
I tried in a docker debian:bulleyes, I dont meet the issue.
 Ubuntu env seems different.

KarimH22, 2025-02-28:
Easy way to reproduce it:

```

docker pull ubuntu:24.04
docker run --rm -it ubuntu:24.04
apt update
install -y python3 build-essential cmake meson pkg-config python3 python3-pip  libncurses-dev libusb-1.0-0-dev python3-mako python3-numpy python3-requests python3-ruamel.yaml python3-setuptools meson python3-pyelftools ninja-build libnuma-dev git libbsd-dev libnl-3-dev libnl-route-3-dev libboost-all-dev
cmake --version
#cmake version 3.28.3
cd /tmp
git clone https://github.com/EttusResearch/uhd.git
git clone http://dpdk.org/git/dpdk
cd dpdk
mkdir /opt/dpdk/
meson setup --prefix=/opt/dpdk build
cd build
ninja -j $(nproc --all) 
ninja install 
ls /opt/dpdk/lib/x86_64-linux-gnu/pkgconfig/
# pc file is present 
cd /tmp/uhd 
git checkout v4.8.0.0
cd host 
mkdir build 
cd build 
PKG_CONFIG_PATH=/opt/dpdk/lib/x86_64-linux-gnu/pkgconfig/ cmake .. 
make -j $(nproc --all)
```

KarimH22, 2025-02-28:
Resolution  : do not use cmake 3.28.3

KarimH22, 2025-03-01:
I reopened the issue because i met it within version cmake 3.31.6