EttusResearch/uhd #841: RFNoC GNURadio Companion Integration / rfnoc_modtool

SJ-Innovation, 2025-03-06:
As referenced in #728 rfnoc_modtool has not been updated to support generation of GRC compatible RFNoC blocks, and the RFNoC blocks it does generate are quite flawed rendering the tool useless/dangerous on established codebases. 

This effectively renders the creation of new RFNoC blocks in UHD 4.8 a manual process, and a bit of a nightmare of one for those of us less intimately familar with the inner working who are wanting to use RFNoC as a tool to get blocks running on USRP devices.

I'm wondering how high a priority getting these tools updated/ functional is or whether we're likely to be stuck like this for a while? 

- SJ

mbr0wn, 2025-06-23:
Hi @SJ-Innovation and everyone else,

as I've mentioned before on the list and here, we are continuously updating our tooling, including `rfnoc_modtool`. Just today we merged support for adding GR C++ blocks, and support for adding GRC bindings is also merged (needs to go public, look out for an update to master in the next days). But just the number of commits on the RFNoC tooling compared to any other single sub-component of UHD is a good indicator, I would say, that these are more than empty promises.

One problem with working on these tools is that they can always be improved, right now you still have to manually edit YAML files to make all the changes exactly as you wish, but that's also hard to automate, and maybe even something that LLM based tools are better at doing than our standard template-based code generation. But the basics are there, which is why I'll be closing this.

A comment on your initial paragraph: You say the tooling can be dangerous on existing codebases. I'm not sure which danger you're referring to here, and I would like to make sure that people don't read this and take something away that is potentially misleading. Can you give some examples where it's dangerous, and what the dangers are?

closed 2025-06-23 by mbr0wn

SJ-Innovation, 2025-06-24:
Thanks for the update & the progress! 

Yes the "dangerous to existing codebases" comment was aimed at the behaviour I noted when adding blocks to existing modules. On several occasions the modtool would overwrite unrelated files in the same directory as (but unrelated to) the generated files for new modules. This was especially prevailant with cmake files, whereby sometimes it would overwrite references to other files when adding modules. It was quite simple to work around by just committing to our OOT git repo (and committing only the stuff we expected to change), but was frustrating until we realised what was going on. 

I imagine this will get sorted as a product of the tools maturing over time :)