EttusResearch/uhd #843: The regex pattern fails to match NMEA 0183 messages

lyfesho, 2025-03-12:
Hello,
I am using a USRP B210 with UHD 4.8.0 installed on Windows 10, and I encountered the following warnings during operation.

After reviewing the source code from host/lib/usrp/gps_ctrl.cpp, I noticed that the regular expression:
`"^\\$GP.*,\\*[0-9A-F]{2}$"`
does not conform to the NMEA 0183 protocol specification (referenced in [GeoS NMEA Protocol v4.0](https://geostar-navi.com/files/docs/geos5/GeoS_NMEA_protocol_v4_0_eng.pdf)).
Specifically, the $GPRMC message format is defined as:
`$--RMC,hhmmss.ss,A,llll.ll,a,yyyyy.yy,a,x.x,x.x,xxxxxx,x.x,a,a,a*hh<CR><LF>`
It seems like the current regex pattern incorrectly requires a comma before the * character, which is not present in the actual NMEA message format.

Here are part of the detailed warning messages I encountered:
C:\Program Files\UHD\lib\uhd\utils>.\query_gpsdo_sensors.exe
Creating the USRP device with: clock_source=gpsdo,time_source=gpsdo...
[INFO] [UHD] Win32; Microsoft Visual C++ version 1942; Boost_108500; UHD_4.8.0.0-release
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Detecting internal GPSDO....
[INFO] [GPS] Found a generic NMEA GPS device
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test...
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz...
[INFO] [B200] Actually got clock rate 16.000000 MHz.
Using Device: Single USRP:
  Device: B-Series Device
  Mboard 0: B210
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: FE-RX2
  RX Channel: 1
    RX DSP: 1
    RX Dboard: A
    RX Subdev: FE-RX1
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: FE-TX2
  TX Channel: 1
    TX DSP: 1
    TX Dboard: A
    TX Subdev: FE-TX1

Setting the reference clock source to "gpsdo"...
Clock source is now gpsdo
Setting the reference clock source to "gpsdo"...
Time source is now gpsdo
Waiting for ref_locked...USRP Locked to Reference.
**************************************Helpful Notes on Clock/PPS Selection**************************************
As you can see, the default 10 MHz Reference and 1 PPS signals are now from the GPSDO.
If you would like to use the internal reference(TCXO) in other applications, you must configure that explicitly.
****************************************************************************************************************
Waiting for the GPSDO to warm up...[WARNING] [GPS] gps_ctrl_impl::update_cache(): Malformed GPSDO string: $GPRMC,015447.00,A,2232.67500,N,11356.67353,E,1.168,27.07,120325,,,A*57
[WARNING] [GPS] gps_ctrl_impl::update_cache(): Malformed GPSDO string: $GPVTG,27.07,T,,M,1.168,N,2.164,K,A*00
[WARNING] [GPS] gps_ctrl_impl::update_cache(): Malformed GPSDO string: $GPGSA,A,2,08,04,09,,,,,,,,,,3.86,3.73,1.00*0D
[WARNING] [GPS] gps_ctrl_impl::update_cache(): Malformed GPSDO string: $GPGSV,2,1,08,04,50,228,34,07,23,321,19,08,79,270,36,09,41,276,44*7B
[WARNING] [GPS] gps_ctrl_impl::update_cache(): Malformed GPSDO string: $GPGSV,2,2,08,16,30,045,,27,57,030,16,31,19,123,15,42,51,128,35*76
[WARNING] [GPS] gps_ctrl_impl::update_cache(): Malformed GPSDO string: $GPGLL,2232.67500,N,11356.67353,E,015447.00,A,A*6B
[WARNING] [GPS] gps_ctrl_impl::update_cache(): Malformed GPSDO string: $GPZDA,015447.00,12,03,2025,00,00*60

The GPSDO is warmed up and talking.
GPS Locked[WARNING] [GPS] gps_ctrl_impl::update_cache(): Malformed GPSDO string: $GPRMC,015448.00,A,2232.67525,N,11356.67370,E,1.125,26.95,120325,,,A*5D
[WARNING] [GPS] gps_ctrl_impl::update_cache(): Malformed GPSDO string: $GPRMC,015450.00,A,2232.67540,N,11356.67410,E,1.047,26.72,120325,,,A*5A
[WARNING] [GPS] gps_ctrl_impl::update_cache(): Malformed GPSDO string: $GPGLL,2232.67540,N,11356.67410,E,015450.00,A,A*69
[WARNING] [GPS] gps_ctrl_impl::update_cache(): Malformed GPSDO string: $GPZDA,015450.00,12,03,2025,00,00*66
Error: ValueError: get_time: Timeout after no valid message found`

mbr0wn, 2026-11-16:
@lyfesho Which GPSDO are you using?

mbr0wn, 2026-11-16:
Anyway, this has been fixed on the latest UHD release.

closed 2026-11-16 by mbr0wn