EttusResearch/uhd #844: uhd-4.8.0.0 compilation fails with gcc-15

yarda, 2025-03-13:
The gcc-15 defaults to the C23 standard. This compiler is the default in Fedora 42 and newer (e.g. rawhide / f43). The C23 introduced new keyword "bool" which (beyond others) causes compilation error:
```
In file included from uhd_dump.c:25:
uhd_dump.h:53:23: error: ‘bool’ cannot be defined via ‘typedef’
   53 | typedef unsigned char bool;
      |                       ^~~~
uhd_dump.h:53:23: note: ‘bool’ is a keyword with ‘-std=c23’ onwards
uhd_dump.h:53:1: warning: useless type name in empty declaration
   53 | typedef unsigned char bool;
      | ^~~~~~~
```
Possibilities are rename, use of `csdtdbool` or explicitly compile with the `-std=gnu17`. In fedora we now temporally compile it with the `-std=gnu17`. There are some other minor problems for which I will open PR.

mbr0wn, 2025-03-14:
`uhd_dump` doesn't even use the `bool` it typedefs. I'll remove that line. Although to be honest, maybe we'll just remove that tool entirely.

yarda, 2025-03-14:
OK, thanks for the info. I will drop the line and retry, they may be more gcc-15 problems, in such case I will update the #845 or create new one.

mbr0wn, 2025-03-14:
@yarda Many thanks for your insights. I'll try and get my hands on a gcc-15 distro (you're probably on rawhide, right?) and see if I can reproduce some of the efforts, but it'll be a few days before I get around to that. To save you some time, here's some comments:
- #845 is good and we've already internally approved it, it's just doing some rounds of CI checking. So just leave it as-is and make a new PR.
- I've also included the `typedef bool` line in our changeset, but overall, don't worry about `uhd_dump`. It's time we retired it. You can assume it's going away.
- If you find some issues within regular UHD that we should know about, those are the most relevant for now.

Again, many thanks for your efforts!

mbr0wn, 2025-03-14:
Actually, I did have a quick look and found some warnings outside of your fixes, but no more errors (at least in UHD itself).

mbr0wn, 2025-03-14:
...the warnings will be fixed because we internally test UHD with `-Werror` (at least with gcc).

yarda, 2025-03-14:
With the dropped typedef bool and the #845 it compiled OK. I am using Fedora 43 (Rawhide), but the same compiler should be in the Fedora 42.

In the #845 I added `cstdint` because the headers used `uint32_t` and gcc-15 is more strict and failed because the type was undefined.

mbr0wn, 2025-03-14:
Excellent, thanks for the feedback. Like I said, I'll remove the remaining warnings while we're at it, then we should be fine for this gcc version.

yarda, 2025-03-14:
One more problem noted, it's probably due to the cmake-4.0 in the Rawhide. On s390x and ppc64le it now installs udev rules to the /usr/lib/uhd/... instead of the  /usr/lib64/uhd as it does on x86_64 (and did previously). But I think it should install the rules to /usr/lib/udev/rules.d/.

yarda, 2025-03-14:
> One more problem noted, it's probably due to the cmake-4.0 in the Rawhide. On s390x and ppc64le it now installs udev rules to the /usr/lib/uhd/... instead of the /usr/lib64/uhd as it does on x86_64 (and did previously). But I think it should install the rules to /usr/lib/udev/rules.d/.

It's probably this one:
https://bugzilla.redhat.com/show_bug.cgi?id=2351130
With the -DLIB_SUFFIX it compiled OK on all arches.

mbr0wn, 2025-04-28:
@yarda I hope to close this out soon, but one FYI: I am planning to start using `GNUInstallDirs` in the UHD CMake, which will make `LIB_SUFFIX` redundant (at least, I hope). Any objections from your end?

yarda, 2025-04-29:
> [@yarda](https://github.com/yarda) I hope to close this out soon, but one FYI: I am planning to start using `GNUInstallDirs` in the UHD CMake, which will make `LIB_SUFFIX` redundant (at least, I hope). Any objections from your end?

NP, several projects I am maintaining downstream already switched to `GNUInstallDirs` and it works. If you need some testing in Fedora, feel free to let me know.