EttusResearch/uhd #847: rfnoc_image_builder does not assert

SJ-Innovation, 2025-03-26:
Assertions should be used judiciously throughout HDL to stop synthesis if a particular piece of IP has unfeasible design parameters or constraints applied. These are often calculated at synth/impl time much like a static assertion in C. For some reason (god only knows why) Vivado DISABLES these assertions by default unless they are forced on. [amd forum](https://adaptivesupport.amd.com/s/question/0D52E00006iHqWDSA0/vhdl-vivado-does-not-check-asserts-in-the-entitystatementpart-during-instantiation?language=en_US)

I'll create a PR that enables these in synth_design.

wordimont, 2025-03-31:
@SJ-Innovation I agree, this should be enabled by default. Was there a specific IP where this was occurring? As far as I know, this setting only affects assertions in VHDL and does not work with SystemVerilog assertions. But, SystemVerilog elaboration system tasks are supported in Vivado regardless of this setting, so that's the preferred way to do this in SystemVerilog.

SJ-Innovation, 2025-04-02:
FYI this is referenced and closed by #848. This was happening in my custom VHDL IPs which makes heavy use of assertions to validate synthesis-time constructs. As my PR adds, the -assert option enables VHDL assertions in all synthesised IPs including out-of-contexts. Enabling this does not cause any assertion failures in SV or V.

closed 2025-04-03 by wordimont