EttusResearch/uhd #851: Logging no trace

GillesC, 2025-04-09:
Given that there is no discussion pane on GitHub, I pose this question/issue here.

I want to investigate one of the variables printed through the ` UHD_LOG_TRACE` function in cpp. (More particular the traces output in the ad9361_dvice.cpp driver)

I have build UHD 4.8 with the -DUHD_LOG_MIN_LEVEL=1 flag.
I have checked this flag was indeed correctly set by checking both the `uhd_config_info --cxx-flags` and `grep UHD_LOG_MIN_LEVEL CMakeCache.txt` in the build dir.

Both are set to:
`-DUHD_LOG_MIN_LEVEL=1` and ``UHD_LOG_MIN_LEVEL:STRING=1`.

I tried to print both to a file and console, but without success:
```python
import os

os.environ["UHD_LOG"] = "trace,all"
os.environ["UHD_LOG_OPTIONS"] = "file=/tmp/uhd_debug.log"
# os.environ["UHD_LOG_OPTIONS"] = "console"

import uhd

# Trigger the retune
usrp = uhd.usrp.MultiUSRP()
usrp.set_rx_freq(917e6)
usrp.set_tx_freq(917e6)

print("Check /tmp/uhd_debug.log for TRACE logs.")
```

No file is created or nothing is printed to console.

wordimont, 2025-04-10:
Discussions take place on the usrp-users mailing list. https://kb.ettus.com/Mailing_Lists

It looks like you set the log level to "DEBUG", but you want to print a "TRACE" message. So, set the log level to `trace` instead. You can add the CMake option `-DUHD_LOG_MIN_LEVEL=trace` when running CMake to include trace-level log messages. And it looks like you want to print to the console (as opposed to a log file), so you'd also want to set `UHD_LOG_CONSOLE_LEVEL`.

I like to use environment variables so I can change the log level at run time. If you've compiled with trace logging, you can turn on the messages by setting the environment variable `UHD_LOG_CONSOLE_LEVEL` to `trace`.

https://kb.ettus.com/The_UHD_logging_facility

GillesC, 2025-04-10:
Thank you for your answer. 

I read https://files.ettus.com/manual/log_8hpp.html#loghpp_levels and https://files.ettus.com/manual/page_logging.html. But the link provided by you clarifies some things. Maybe it would be better to refer to https://kb.ettus.com/The_UHD_logging_facility in those document as well. Anyhow ;)
I will have to recompile from source in that case (using `-DUHD_LOG_MIN_LEVEL=0`).

closed 2025-04-10 by GillesC