EttusResearch/uhd #857: Help please... Subdevice Selection from RFNoC Rx Radio Block

SJ-Innovation, 2025-05-07:
Hi all,

Changing between frontends using the "USRP Source" block is straightforward using the ChX Subdev Spec box in GNURadio. This only works however for a USRP immediately exporting its samples to GNURadio. 

Having created an RFNoC flowgraph with multiple custom blocks and the RFNoC Rx Radio block I am unable to change which subdev the block connects to. It always seems to default to A:0, whereas I need to connect to A:1. 
How would i go about this? I've trawled documentation & source code, though could do with some external input as i'm not making headway. 

Thanks!
Sam

mbr0wn, 2025-05-07:
@SJ-Innovation please keep these kinds of questions to the mailing list -- this is for bug reports, although we do accept feature requests too.

Subdev specs are basically obsolete with RFNoC. To choose A:1, you simply connect port 1 (instead of port 0) of your radio block. If you want to switch to B:0/1, then set 'instance select' to 1 instead of 0.

closed 2025-05-07 by mbr0wn

SJ-Innovation, 2025-05-07:
I'll be honest i forgot the mailing list even existed & was so irritated id have done a deal with the antichrist to find a solution. Mailing list for future questions, rgr that. 👍

mbr0wn, 2025-05-07:
> I'll be honest i forgot the mailing list even existed & was so irritated id have done a deal with the antichrist to find a solution.

Hopefully your soul is still intact and in your possession.

>  Mailing list for future questions, rgr that. 👍

Thanks!

daisylrgh, 2025-05-23:
Could we please reopen this and turn it into a feature request? Our USRP is not easily accessible and so a feature in which we do not have to manually connect/disconnect would be extremely useful. We have tried your suggestion of "If you want to switch to B:0/1, then set 'instance select' to 1 instead of 0" with no luck. We also cant leave one of the ports open in RFNoC, GNURadio complains. @mbr0wn please refer to this email sent by Sam 

[image]